cncjs/cncjs-pi-raspbian #1: Wireless

nickm324, 2018-07-03:
Whats wrong with the wireless in this image. I cannot get it to connect no matter what I do. has something been disabled?

AustinSaintAubin, 2018-07-03:
Nothing has been disabled. Network was testing over ethernet though.
Take a look at this guide: [raspberrypi.org: SETTING WIFI UP VIA THE COMMAND LINE](https://www.raspberrypi.org/documentation/configuration/wireless/wireless-cli.md)
Is your SSID hidden? Take a look at the "Hidden networks" section of that article if it is.
You might also take a look at [this article](https://weworkweplay.com/play/automatically-connect-a-raspberry-pi-to-a-wifi-network/) as well.  

Also, have you tried using the raspi-config tool:
pi@cncjs:~ $ `sudo raspi-config`

2. Network Options
N2. Wi-fi (then input your SSID & passphrase)

nickm324, 2018-07-03:
In order to get it to work I had to go into raspi-config and in network options I had to ENABLE  predictable network interface names, then it started working. Apparently this is disabled in the image.

AustinSaintAubin, 2018-07-03:
I will get that corrected and post an new image.
Thanks for letting me know.

nickm324, 2018-07-11:
I have been running the image for about a week, so far so good. I do still have a strange issue that sometimes on boot the wireless will not come up and if I shut it down and restart the Pi it will will work again. Not sure whats going on but this seems to happen every other time I boot the pi.

nickm324, 2018-08-23:
Follow up: I have been having constant issues with the Wifi just dropping out randomly and not working without a reboot, I have tried all the suggestions so far about disabling power management onthe  wifi hardware etc. nothing seems to help. I have found some evidence that there is a firmware bug in certain versions of the rasbian distro's and am researching that more to try and isolate it to this version if that is the case. if there is a job running it will continue to run with out issue but access to the web interface is lost once the wifi dies, only option then is to wait for the running job to complete then reboot the Pi.

AustinSaintAubin, 2018-08-23:
I am working on a new version of the cncjs raspbian image, and also a setup script that can preform the setup. Let me know your findings.

nickm324, 2018-08-23:
There is a pretty lengthy thread about  the wifi issue here https://lb.raspberrypi.org/forums/viewtopic.php?t=188891https://lb.raspberrypi.org/forums/viewtopic.php?t=188891

There is even some acknowledgement from some of the designers from raspberry Pi that there is/was a know issue with Cypress firmware and are waiting on feedback form them or something.

There also is some talk about it being a power supply issue, so I am going to try out another PS I have and see if that makes any difference. Will Keep you posted.

AustinSaintAubin, 2018-09-13:
New version released, give it a try.
https://github.com/cncjs/cncjs-pi-raspbian/releases

AustinSaintAubin, 2018-09-29:
@nickm324 You ok if I close this issue? You still have anything related to this issue that is unresolved?