cncjs/cncjs-pi-raspbian #2: Can't Connect to Pi

tigert10, 2018-07-21:
Hi, I have installed this release on a pi and I can connect to the pi via wifi or LAN using PuTTY, I can see the pi connected to the router, however if I try and access it through a browser by typing in its IP address it won't connect.  Any ideas?  Thanks in advance!

tigert10, 2018-07-21:
Just to be clear: when I remote into the pi I see the CNCJS logo and instructions that say: Navigate your web browser to:
        http://cncjs/
        http://192.168.0.94/
and when I try to do that from a Chrome browser it does not connect.  I tried disabling the firewall on the router with no success either.  Any advice is appreciated. Thanks!

tigert10, 2018-07-21:
I tried a fresh install just to double check that I didn't mess something up, and after the fresh install I ONLY changed the wifi settings to connect via wifi.  Then I could PuTTY into the pi but when I try and access CNCjs from a browser it will not connect.  I have tried from a desktop and a couple of mobile devices with no success.

tigert10, 2018-07-21:
So, I have to run
pm2 start $(which cncjs) -- --port 8000
to be able to connect to the pi via another browser

following the setup guide to get it to auto start on boot does not seem to be working.  Any ideas?

tigert10, 2018-07-21:
So I finally figured it out.  Instead of "sudo reboot" after changing the pm2 startup script I unplugged the pi instead. It worked...go figure. It's amazing what a bit of luck will do sometimes.

closed 2018-07-21 by tigert10

AustinSaintAubin, 2018-07-22:
Odd. This image is still in beta, I have a few bugs is seems that need to be worked out. I will see what I can do to debug what is going on.

tigert10, 2018-07-23:
Thanks for your work on this. It will help a lot of people to get over the biggest hurdle of the diy cnc software.