cncjs/cncjs-pi-raspbian #4: Wont start X Server

nickm324, 2018-10-17:
I wrote the 1.0.1 image to SD card booted it with my LCD drivers installed and the display works but it ends up with and error that says xint: unable to connect to X server: connection refused.

Fresh image no changes other than add LCD drivers.

AustinSaintAubin, 2018-10-17:
Was the web kiosk working before LCD driver install?
You might re flash the SD card and test over HDMI before LCD driver install.

If you are using LCD-show drivers you might look at this post; [Rotating Touch [Raspbian 4.14, 3.5 LCD, RP 2]](https://github.com/goodtft/LCD-show/issues/93)

nickm324, 2018-10-17:
No, I wrote the image, booted it (display not working) used ssh to access the Pi, installed my LCD drivers, rebooted, Display now works, it continues to boot and ends with the start X error.

nickm324, 2018-10-17:
If I try and start X manually from the terminal here is what is displayed on the LCD
[https://1drv.ms/u/s!AjC1qsnTdJCfgdQBh1bekCD3ax16HQ](url)

nickm324, 2018-10-21:
I figured it out, all I had to do was create 98-lcd-show.conf  because it did not exist and the below configuration in it. That is not included in the 1.01 release.

# https://www.raspberrypi.org/forums/viewtopic.php?t=162323
# https://www.raspberrypi.org/forums/viewtopic.php?t=148678
cat > /etc/X11/xorg.conf.d/98-lcd-show.conf <<"EOF"
Section "Device" 
	# WaveShare SpotPear 3.5", framebuffer 1
	Identifier "uga" 
	driver "fbdev" 
	Option "fbdev" "/dev/fb1" 
	Option "ShadowFB" "off"
	#Option "Rotate" "UD"  # Flip Upside Down
	Option "SwapbuffersWait" "true"
EndSection 

Section "Monitor" 
	# Primary monitor. WaveShare SpotPear 480x320
	Identifier "WSSP" 
EndSection 

Section "Screen" 
	Identifier "primary" 
	Device "uga" 
	Monitor "WSSP" 
EndSection 

Section "ServerLayout" 
	Identifier "default" 
	Screen 0 "primary" 0 0
EndSection
EOF

AustinSaintAubin, 2018-10-21:
Glad you where able to figure it out. I had wondered if it was the same issue I had with LCD-Show.

closed 2018-10-21 by AustinSaintAubin