cncjs/cncjs-pi-raspbian #5: notifications not working

bipbipbip, 2018-10-21:
hey,

for example if i click shutdown, not notifications windows opening, only a little N on the upper right

AustinSaintAubin, 2018-10-21:
???
Consider importing the template I created, [cncjs-app-1.9.15.json](https://raw.githubusercontent.com/cncjs/cncjs-pi-raspbian/master/cncjs-app-1.9.15.json)
You can do that in the settings area.

bipbipbip, 2018-10-22:
hi, 
thanks for your answer
but i've already done that as explained in your instructions.

i've created a custom command to shutdown (because on your version it don't exist), normally when i click on this command it should have a notification window. but (that's the issue), no notification  appears

closed 2020-09-28 by AustinSaintAubin