cncjs/cncjs-pi-raspbian #6: Include keyboard pendant

dpichevin, 2018-11-23:
Hi,

Any chance we could get a build of this that includes a properly installed keyboard pendant (for the Rii wireless keyboard)?

I've been at it for a couple days and no matter what I try, I cannot properly install anything further on this system. Even after a fresh reflash, any npm install ends in error. Installing of the pendant ends in it not finding the bindings. Installing the node hid, the usb lib, the serial port seems to work after I do a sudo apt-get update (those npm error messages are gone), but in the end, the pendant install still errors out, and it still won't find the bindings. Don't know what else to try, and I'd really love to get this working so I can get to cutting some wood finally. This image was very useful, as the standard CNCjs install did not work, so if it could also have the keyboard pendant, that would be really swell :)

Thanks.

AustinSaintAubin, 2018-11-24:
I will release an updated version with all the pendants installed soon (might be a few days), per your request.

dpichevin, 2018-11-24:
Awesome! I discovered the little keyboard icon at the top right to jog with the arrow keys but it still lacks a bunch of key functions.

dpichevin, 2019-01-12:
Still very interested in this, when you get a chance :)

dpichevin, 2019-01-15:
Actually, I think that I'm switching to the Windows version at this point, as it's more practical and seems to have less input lag. Had an older Windows laptop waiting to be used, and it's definitely the easiest way to get a working CNCjs install.

birthstar, 2019-05-07:
Hi also finding the keyboard pendant a challenge to install on the Raspberry PI. If it is possible to create a image that has it working that would be great.

AustinSaintAubin, 2020-09-28:
Will look at adding more pendants to the new install script in the future.

closed 2020-09-28 by AustinSaintAubin