cncjs/cncjs-pi-raspbian #11: Invalid environment assignment

weyersma, 2020-10-17:
I am trying to use the script on a Pi4. I have tried different images of Raspberry Pi OS and get the same outcome. When the service starts the following messages are displayed:

  Status of Service - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    ● cncjs.service - CNCjs is a full-featured web-based interface for CNC controllers running Grbl, Marlin, Smoothieware, or TinyG.
   Loaded: loaded (/etc/systemd/system/cncjs.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-10-17 14:42:51 SAST; 2s ago
     Docs: https://github.com/cncjs/cncjs
 Main PID: 15273 (node)
    Tasks: 7 (limit: 1433)
   CGroup: /system.slice/cncjs.service
           └─15273 node /usr/bin/cncjs

Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: --port
Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: 8000
Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: --config
Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: /home/pi/.cncjs/cncrc.cfg
Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: --watch-directory
Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: /home/pi/.cncjs/watch
Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: --mount
Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: /tinyweb:/home/pi/.cncjs/pendant-tinyweb/src
Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: --mount
Oct 17 14:42:53 raspberrypi systemd[1]: /etc/systemd/system/cncjs.service:18: Invalid environment assignment, ignoring: /tablet:/home/pi/.cncjs/pendant-shopfloor-tablet/src
  - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -

AustinSaintAubin, 2020-11-01:
Looking into this issue. Sorry for the late reply.

AustinSaintAubin, 2020-11-01:
Looks like issue with service encapsulation. I will take a look at install script and make sure us constrained.

AustinSaintAubin, 2020-11-01:
Try now with, script updated to v1.0.22 to fix service issues.

AustinSaintAubin, 2020-11-22:
Assuming this is resolved.

closed 2020-11-22 by AustinSaintAubin