cncjs/cncjs-pi-raspbian #12: Install appears complete, "cncjs" not found

AcrimoniousMirth, 2020-10-18:
So for a week or so I've been trying to install CNCJS as used it two years back and loved it. Can't get past a missing NP2 environment file issue which none of the guidance has managed to fix so been attempting to install via this image instead, being somewhat at my wit's end.
Anyway, installation looks like its gone through alright but then it refuses to recognise or open CNCJS. Here's my printout. Any help/advice would be greatly appreciated, I don't remember it being this tough last time :(

```
pi@CNCPi:~ $ curl -sSL https://raw.githubusercontent.com/cncjs/cncjs-pi-raspbian/master/cncjs_install.sh | bash
CNCjs Installer v1.0.21(2020-10-17) by: Austin St. Aubin
     _______   ________  _
    / ____/ | / / ____/ (_)____
   / /   /  |/ / /     / / ___/
  / /___/ /|  / /___  / (__  )
  \____/_/ |_/\____/_/ /____/
                  /___/

  Installing CNCjs
   - https://cnc.js.org
   - https://github.com/cncjs
   
  CNCjs is a full-featured web-based interface for CNC controllers running Grbl, Marlin, Smoothieware, or TinyG.
  For a more complete introduction, see the Introduction section of the wiki page ( https://github.com/cncjs/cncjs/wiki/Introduction ).
  NOTE: This installer logs to syslog. You can view the syslog, with terminal command 'tail -f -n 50 /var/log/syslog'
  ==========================================================================
  [i]  Detected OS: [ raspbian | 10 | Compatible GUI ]   
  [✓]  Detected OS is compatable with this install script.   
  [✓]  Updating System Packages   
  [✓]  Upgrading System Packages (this can take a while, please wait)   
  [✓]  Upgrading System Distribution (this can take a while, please wait)   
  [✓]  Fixing Broken Packages (if any)   

  Setup Node.js & NPM via Package Manager   
  [✓]  Installing Node.js v10.x Package Source   
  [✓]  Installing Node.js v10.x via Package Manager   
  [✓]  Installing Build Essential   
  [✓]  Installing Latest Node Package Manager (NPM)   
  [i]  NodeJS: 	v14.13.1 	|	 /usr/bin/node   
  [i]  NPM: 	v6.14.8 	|	 /usr/bin/npm   

  Install CNCjs   
  [✓]  Installing CNCjs (v1.9.22) with NPM   
pi@CNCPi:~ $ sudo reboot
sudo: unable to resolve host CNCPi: Name or service not known
Connection to 192.168.0.54 closed by remote host.
Connection to 192.168.0.54 closed.
```
At this point I try to navigate to 192.168.0.1:8000 via Safari but it just tells me it can't connect. So I attempt to ssh in and see if there's anything I can do.
```
MBP:~ user$ ssh pi@192.168.0.54
pi@192.168.0.54's password: 
Linux CNCPi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Oct 18 19:39:19 2020
/home/pi/.nvm/versions/node/v4.9.1/lib/node_modules/npm/bin/npm-cli.js:87
      let notifier = require('update-notifier')({pkg})
      ^^^

SyntaxError: Block-scoped declarations (let, const, function, class) not yet supported outside strict mode
    at exports.runInThisContext (vm.js:53:16)
    at Module._compile (module.js:373:25)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:140:18)
    at node.js:1043:3
nvm is not compatible with the npm config "prefix" option: currently set to ""
Run `npm config delete prefix` or `nvm use --delete-prefix v4.9.1 --silent` to unset it.
pi@CNCPi:~ $ cncjs -V
-bash: cncjs: command not found
pi@CNCPi:~ $ cncjs
-bash: cncjs: command not found
pi@CNCPi:~ $ sudo cncjs
sudo: unable to resolve host CNCPi: Name or service not known
sudo: cncjs: command not found
```

Thanks for the help, I was really hoping to get my CNC up and running but its been rather delayed. I'd like to sort out all the controls before wiring up.

AustinSaintAubin, 2020-11-01:
Looking into this issue. Sorry for the late reply.

AustinSaintAubin, 2020-11-01:
Looks like your install might have error out quietly. 
Try again with refesh install. Run this command `tail -f -n 50 /var/log/syslog` and post output as well.

AustinSaintAubin, 2020-11-01:
Try now with, script updated to v1.0.22 to fix service issues.

AustinSaintAubin, 2020-11-22:
Assuming this is resolved.

closed 2020-11-22 by AustinSaintAubin

AcrimoniousMirth, 2020-11-23:
Sorry, didn’t get notified of your comments over all this time. Will try this this afternoon, thanks! 
I loved CNCjs two years back, would be amazing to use it again. UGS just not cutting it.

AcrimoniousMirth, 2020-11-24:
> Looks like your install might have error out quietly.
> Try again with refesh install. Run this command `tail -f -n 50 /var/log/syslog` and post output as well.

Sorry for delay again, just ran with the updated install (thank you for that). I also ran that common afterwards.
`pi@CNCPi:~ $ tail -f -n 50 /var/log/syslog
Nov 24 19:15:12 CNCPi systemd[1]: Mounted FUSE Control File System.
Nov 24 19:15:13 CNCPi dhcpcd[451]: wlan0: leased 192.168.0.54 for 864000 seconds
Nov 24 19:15:13 CNCPi avahi-daemon[340]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.0.54.
Nov 24 19:15:13 CNCPi avahi-daemon[340]: New relevant interface wlan0.IPv4 for mDNS.
Nov 24 19:15:13 CNCPi avahi-daemon[340]: Registering new address record for 192.168.0.54 on wlan0.IPv4.
Nov 24 19:15:13 CNCPi dhcpcd[451]: wlan0: adding route to 192.168.0.0/24
Nov 24 19:15:13 CNCPi dhcpcd[451]: wlan0: adding default route via 192.168.0.1
Nov 24 19:15:13 CNCPi pm2[479]: [PM2] PM2 Successfully daemonized
Nov 24 19:15:13 CNCPi pm2[479]: [PM2] Resurrecting
Nov 24 19:15:13 CNCPi pm2[479]: [PM2] Restoring processes located in /home/pi/.pm2/dump.pm2
Nov 24 19:15:14 CNCPi pm2[479]: ┌─────┬───────────┬─────────────┬─────────┬─────────┬──────────┬────────┬──────┬───────────┬──────────┬──────────┬──────────┬──────────┐
Nov 24 19:15:14 CNCPi pm2[479]: │ id  │ name      │ namespace   │ version │ mode    │ pid      │ uptime │ ↺    │ status    │ cpu      │ mem      │ user     │ watching │
Nov 24 19:15:14 CNCPi pm2[479]: └─────┴───────────┴─────────────┴─────────┴─────────┴──────────┴────────┴──────┴───────────┴──────────┴──────────┴──────────┴──────────┘
Nov 24 19:15:14 CNCPi systemd[1]: Started PM2 process manager.
Nov 24 19:15:14 CNCPi systemd[1]: Reached target Multi-User System.
Nov 24 19:15:14 CNCPi systemd[1]: Reached target Graphical Interface.
Nov 24 19:15:14 CNCPi systemd[1]: Starting Update UTMP about System Runlevel Changes...
Nov 24 19:15:14 CNCPi systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Nov 24 19:15:14 CNCPi systemd[1]: Started Update UTMP about System Runlevel Changes.
Nov 24 19:15:14 CNCPi systemd[1]: Startup finished in 3.986s (kernel) + 16.161s (userspace) = 20.147s.
Nov 24 19:15:14 CNCPi systemd[1]: systemd-rfkill.service: Succeeded.
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Activating via systemd: service name='org.gtk.vfs.UDisks2VolumeMonitor' unit='gvfs-udisks2-volume-monitor.service' requested by ':1.7' (uid=1000 pid=767 comm="pcmanfm --desktop --profile LXDE-pi ")
Nov 24 19:15:15 CNCPi systemd[580]: Starting Virtual filesystem service - disk device monitor...
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Successfully activated service 'org.gtk.vfs.UDisks2VolumeMonitor'
Nov 24 19:15:15 CNCPi systemd[580]: Started Virtual filesystem service - disk device monitor.
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Activating via systemd: service name='org.gtk.vfs.AfcVolumeMonitor' unit='gvfs-afc-volume-monitor.service' requested by ':1.7' (uid=1000 pid=767 comm="pcmanfm --desktop --profile LXDE-pi ")
Nov 24 19:15:15 CNCPi systemd[580]: Starting Virtual filesystem service - Apple File Conduit monitor...
Nov 24 19:15:15 CNCPi gvfs-afc-volume-monitor[970]: Volume monitor alive
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Successfully activated service 'org.gtk.vfs.AfcVolumeMonitor'
Nov 24 19:15:15 CNCPi systemd[580]: Started Virtual filesystem service - Apple File Conduit monitor.
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Activating via systemd: service name='org.gtk.vfs.GoaVolumeMonitor' unit='gvfs-goa-volume-monitor.service' requested by ':1.7' (uid=1000 pid=767 comm="pcmanfm --desktop --profile LXDE-pi ")
Nov 24 19:15:15 CNCPi systemd[580]: Starting Virtual filesystem service - GNOME Online Accounts monitor...
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Successfully activated service 'org.gtk.vfs.GoaVolumeMonitor'
Nov 24 19:15:15 CNCPi systemd[580]: Started Virtual filesystem service - GNOME Online Accounts monitor.
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Activating via systemd: service name='org.gtk.vfs.MTPVolumeMonitor' unit='gvfs-mtp-volume-monitor.service' requested by ':1.7' (uid=1000 pid=767 comm="pcmanfm --desktop --profile LXDE-pi ")
Nov 24 19:15:15 CNCPi systemd[580]: Starting Virtual filesystem service - Media Transfer Protocol monitor...
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Successfully activated service 'org.gtk.vfs.MTPVolumeMonitor'
Nov 24 19:15:15 CNCPi systemd[580]: Started Virtual filesystem service - Media Transfer Protocol monitor.
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Activating via systemd: service name='org.gtk.vfs.GPhoto2VolumeMonitor' unit='gvfs-gphoto2-volume-monitor.service' requested by ':1.7' (uid=1000 pid=767 comm="pcmanfm --desktop --profile LXDE-pi ")
Nov 24 19:15:15 CNCPi systemd[580]: Starting Virtual filesystem service - digital camera monitor...
Nov 24 19:15:15 CNCPi dbus-daemon[605]: [session uid=1000 pid=605] Successfully activated service 'org.gtk.vfs.GPhoto2VolumeMonitor'
Nov 24 19:15:15 CNCPi systemd[580]: Started Virtual filesystem service - digital camera monitor.
Nov 24 19:15:20 CNCPi dhcpcd[451]: wlan0: no IPv6 Routers available
[... cut ...]
Nov 24 19:15:27 CNCPi vncserver-x11[487,root]: SPrintConnMgr: Failed to add printer: server-error-service-unavailable
Nov 24 19:15:33 CNCPi systemd[1]: systemd-fsckd.service: Succeeded.
Nov 24 19:15:46 CNCPi systemd-timesyncd[307]: Synchronized to time server for the first time 217.114.59.66:123 (2.debian.pool.ntp.org).
Nov 24 19:15:51 CNCPi systemd[1]: systemd-hostnamed.service: Succeeded.
Nov 24 19:17:01 CNCPi CRON[1285]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
`

Tried navigating to the portal (192.168.0.54:8000) but couldn't reach it. Probably missing something obvious, I'm very tired!