cncjs/cncjs-pi-raspbian #13: Install Complete: No Available Ports

DerrickLamers, 2020-11-03:
CNCjs is not seeing my serial connection. I have tried all serial ports, rebooted several times, and did to fresh installs but the serial connection is never found.

DerrickLamers, 2020-11-04:
Update, if the stop the cncjs service and run cncjs manually, it works normal.

AustinSaintAubin, 2020-11-07:
Will have to do some more digging into systemd. Could be a few different issues. Thanks for report.

AustinSaintAubin, 2020-11-08:
Fixed in latest version script. Looks like it was issue with service path and node serial package.
Let me know if still having issues.

AustinSaintAubin, 2020-11-22:
Assuming this is resolved.

closed 2020-11-22 by AustinSaintAubin

TheEagle13, 2020-12-17:
I just did a fresh install using this script, and I have the same issue... "No Ports Available".

My CNC is connected as `/dev/ttyUSB0`