cncjs/cncjs-pi-raspbian #14: kiosk mode chromium

jelinj8, 2020-11-09:
As the kiosk mode browser is starting as root, the screen shows dialog about waiting for cncJS, then it stays black. Adding --no-sandbox option to chromium switches solves this.

AustinSaintAubin, 2020-11-10:
Will add this flag to chrome.

AustinSaintAubin, 2020-11-14:
Fixed how kiosk launches, runs under current user now. Please let me know if does not work.

zmrdko, 2021-01-06:
hello, I have issues with kiosk mode. In setup I will set http://localhost:8000/tablet or http://localhost:8000/tinyweb but after reboot, I get default reboot.
I am using latest fresh raspbian lite installation.

jelinj8, 2021-01-06:
I had this problem too, but not only in kiosk mode, even external browsers are redirected to main page, so I think it isn't related to this problem.

zmrdko, 2021-01-06:
using browser I am able to get to particular pendant

AustinSaintAubin, 2021-01-12:
Looking into this shortly.

AustinSaintAubin, 2021-01-17:
Looks like this is resolved in latest script.

closed 2021-01-17 by AustinSaintAubin