cncjs/cncjs-pi-raspbian #15: FYI: how to "--allow-remote-access" or how to have a web site access your cncjs server if your not on your own network.

graphtjb, 2020-11-20:
on your raspberry pi do something like this. to find systemd.

`cd /etc/systemd/system/ `

It is a suggestion that you read systemd and understand it as it is yet another way to make execute code on boot.
If you want skip all that.. aka "that show me the money" aka "bro you know i don't need all that"

then do something to the effect as follows.

`sudo nano /etc/systemd/system/cncjs.service`

I replaced the line 
`ExecStart=/usr/bin/cncjs --port 8000 --config "/home/pi/.cncjs/cncrc.cfg" --watch-directory "/home/pi/.cncjs/watch" --mount /tinyweb:/home/pi/.cncjs/pendant-tinyweb/src --mount /tablet:/home/pi/.cncjs/pendant-shopfloor-tablet/src --mount /widget-boilerplate:https://cncjs.github.io/cncjs-widget-boilerplate/v2/`

with this line (note the word "--allow-remote-access"
`ExecStart=/usr/bin/cncjs --port 8000 --config "/home/pi/.cncjs/cncrc.cfg" --allow-remote-access --watch-directory "/home/pi/.cncjs/watch" --mount /tinyweb:/home/pi/.cncjs/pendant-tinyweb/src --mount /tablet:/home/pi/.cncjs/pendant-shopfloor-tablet/src --mount /widget-boilerplate:https://cncjs.github.io/cncjs-widget-boilerplate/v2/`

I installed a few more things but all you need is the word --allow-remote-access on the line with "ExecStart"
I would suggest digging through more code. I got to their server and found white lists and black lists. aka (lists that are inclusive and lists that exclusive.)(Bob almighty don't be rule nazi then say code racist.) I would suggest this and look to use mac address filtering. so long as you have just known device allowed.. you can for get most of the net as they know not this spoofing of the big mac. I am just saying it sound more secure.. and limiting the password attempts per day and the total amount password attempts per day. Am i nagging the net denizens of password security? yes.. yes i am. tsk bro you know i got you back but you can not be that dumb.

AustinSaintAubin, 2020-11-23:
Having a little trouble interpreting. Are you requesting I add a flag for `--allow-remote-access` during the install process?

graphtjb, 2020-11-23:
perhaps there is a better way to give fellow users advice. This is your project and i was looking to post information i thought was useful. I was looking to leave the solution to the problem i had. I did not know how you had cncjs execute on start up or boot up. (you used systemd(maybe you could have used cron)). I found the answer and thought i would post this information should it help another user.  I was typing quickly and perhaps i should have had a more professional attitude to keep my post to a minimum. I was going to extend my post when i had my Mac filtering going and was better at white lists and black lists. I was also looking to show how to use your install to extend the cncjs platform via a plug in. So a for example i did this to make a plugin in addition to this wiki for cncjs. perhaps there is a better place to put this information?

AustinSaintAubin, 2020-11-26:
Documenting by use of an issue is good. I appreciate your input, and made changes to the code because of your input.
Thank you, your comments indicated the chrome task was running as root, which it should have have been. I fixed the related bug causing this issue, and the code is better because of your input calling out the issue. Thank you again.

Though, I recommend keeping comments as minimal and concise as possible in the future.
Keep in-mind as well, that you also have to be mindful of others of all backgrounds and understandings, some who might be using google translate. Simple and concise usually translates well, especially from english to english.

Another thing, though this really is a side note. Formatting matters. 
Think of text just as much graphical as informational. Structure, when referring to type is a [Double Entendre](https://en.wikipedia.org/wiki/Double_entendre).
Have a read of: [The Non-Designer's Design Book by Robin Williams](https://www.google.com/books/edition/The_Non_designer_s_Design_Book/pqHfBQAAQBAJ

closed 2020-11-26 by AustinSaintAubin