cncjs/cncjs-pi-raspbian #16: LCD3.5 RPI3

RemyFr, 2020-11-21:
Hello,

First thank you for your fantastic work.

I am trying to install CNCJS in my RPI3 to use it with a GPIO LCD 3.5 (the display works fine).

I followed the instructions (install raspbian lite and the CNCJS), during the installation i choosed in the menu TinyWebPendant, after rebooting CNCJS worked (no curser for the mouse). Then i install the display with the console, reboot, LCD detected but CNCJS dont boot. Here a photo of my screen.

[image]

Have you issues ? Did i made something wrong  ?

Sorry for my bad english.
Rémy

AustinSaintAubin, 2020-11-22:
The no cursor issue is because I left the `nocursor` flag in the `startx -- -nocursor` command of ~/.bashrc
https://github.com/cncjs/cncjs-pi-raspbian/blob/1e2238fa9d124ca58741b92d13f64e33666dbe2e/cncjs_install.sh#L798
I will make this optional in future release, for know you can remove '-- -nocursor' from the Start Web Kiosk line `[[ -z $DISPLAY && $XDG_VTNR -eq 1 ]] && startx -- -nocursor` in the `~/.bashrc` file.

------------

The UI not booting could be an issue with GPIO LCD 3.5 screen. 
Please send me a link to the screen and script you ran for the screen, I will try to test screen.

RemyFr, 2020-11-23:
Thanks for the replie.

My lcd is an old version of Landzo product (can't link).

The script i used  : 

sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show

Regards,

Remy

RemyFr, 2020-11-23:
If this can help you i have a photo  :
[image]

AustinSaintAubin, 2020-11-25:
Cool, I have one of similar build and will test.

AustinSaintAubin, 2020-11-26:
Testing with the following on Pi2B (all I have thats not in use at the moment):

1. Installed [3.5inch RPi LCD](https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A))
```
git clone https://github.com/waveshare/LCD-show.git
cd LCD-show/
chmod +x LCD35-show
 ./LCD35-show
```
2. Ran the [CNCjs Install Script](https://github.com/cncjs/cncjs-pi-raspbian).

RemyFr, 2020-12-07:
Hello,

Sorry for the late reply,

Your script for the LCD works (i think it is better than the first i used) however i have the same errors displayed.

Regards.

AustinSaintAubin, 2021-01-17:
Just testing with the latest scrip. Seems like everything worked.

Try:
1. Install w/ cncjs-pi-raspbian script
2. Install Screen with commands noted below.
```
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show
```

closed 2021-03-24 by AustinSaintAubin