cncjs/cncjs-pi-raspbian #17: Install CNCjs failed w/ error 1

AcrimoniousMirth, 2020-12-28:
Merry Christmas and a happy new year!
Tried to install via the script on RPi 3B, here's my entire readout. I think I'm having ongoing issues with NPM for some reason...

Upon opening Terminal:
`/home/pi/.nvm/versions/node/v4.9.1/lib/node_modules/npm/bin/npm-cli.js:87
      let notifier = require('update-notifier')({pkg})
      ^^^

SyntaxError: Block-scoped declarations (let, const, function, class) not yet supported outside strict mode
    at exports.runInThisContext (vm.js:53:16)
    at Module._compile (module.js:373:25)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:140:18)
    at node.js:1043:3
nvm is not compatible with the npm config "prefix" option: currently set to ""
Run `npm config delete prefix` or `nvm use --delete-prefix v4.9.1 --silent` to unset it.`

Proceeding with install.
`pi@CNCPi:~ $ curl -sSL https://raw.githubusercontent.com/cncjs/cncjs-pi-raspbian/master/cncjs_install.sh | bash
CNCjs Installer v1.1.2(2020-11-11) by: Austin St. Aubin
     _______   ________  _
    / ____/ | / / ____/ (_)____
   / /   /  |/ / /     / / ___/
  / /___/ /|  / /___  / (__  )
  \____/_/ |_/\____/_/ /____/
                  /___/

  Installing CNCjs
   - https://cnc.js.org
   - https://github.com/cncjs
   
  CNCjs is a full-featured web-based interface for CNC controllers running Grbl, Marlin, Smoothieware, or TinyG.
  For a more complete introduction, see the Introduction section of the wiki page ( https://github.com/cncjs/cncjs/wiki/Introduction ).
  NOTE: This installer logs to syslog. You can view the syslog, with terminal command: tail -f -n 50 /var/log/syslog 
  ==========================================================================
  [i]  Detected OS: [ raspbian | 10 | Compatible GUI ]   
  [✓]  Detected OS is compatable with this install script.   
  [✓]  Updating System Packages  
  [✓]  Upgrading System Packages (this can take a while, please wait)  
  [✓]  Upgrading System Distribution (this can take a while, please wait)  
  [✓]  Fixing Broken Packages (if any)  

  Setup Node.js & NPM via Package Manager   
  [✓]  Installing Node.js v10.x Package Source  
  [✓]  Installing Node.js v10.x via Package Manager  
  [✓]  Installing Build Essential  
  [✓]  Installing Latest Node Package Manager (NPM)  
  [i]  NodeJS: 	v14.13.1 	|	 /usr/bin/node   
  [i]  NPM: 	v6.14.10 	|	 /usr/bin/npm   `

This is the first failure.
`  Install CNCjs   
  [✗]  Installing CNCjs (v1.9.22) with NPM | Error Code: 1  
   └── Try to re-run this part of the script after rebooting.
  Latest Syslog Entries - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    Dec 28 16:21:51 CNCPi bash: npm ERR! 
Dec 28 16:21:51 CNCPi bash: npm ERR! Failed at the @serialport/bindings@2.0.8 install script.
Dec 28 16:21:51 CNCPi bash: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Dec 28 16:21:52 CNCPi bash: 
Dec 28 16:21:52 CNCPi bash: npm ERR! A complete log of this run can be found in:
Dec 28 16:21:52 CNCPi bash: npm ERR!     /root/.npm/_logs/2020-12-28T16_21_52_037Z-debug.log
  - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 
  [✓]  Set User TTY Permissions  

  Download & Install CNCjs Pendants & Widgets	[ /home/pi/.cncjs ]   
  [✓]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch	( /home/pi/.cncjs )  
  [⣾]  Download & Install: Pendant Shopfloor Tablet	( http://192.168.0.54/ta  [⣽]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣻]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⢿]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⡿]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣟]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣯]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣷]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣾]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣽]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣻]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⢿]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⡿]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣟]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣯]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [⣷]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/ta  [✓]  Download & Install: Pendant Shopfloor Tablet-tabl( http://192.168.0.54/tablet )		[ /home/pi/.cncjs/pendant-shopfloor-tablet ]  
[... 35 lines cut ...]
Dec 28 16:14:21 CNCPi systemd[1]: /etc/systemd/system/cncjs.service:19: Executable "-port" not found in path "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin", ignoring
Dec 28 16:14:21 CNCPi systemd[1]: cncjs.service: Service has no ExecStart=, ExecStop=, or SuccessAction=. Refusing.
Dec 28 16:14:22 CNCPi systemd[1]: /etc/systemd/system/cncjs.service:19: Executable "-port" not found in path "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin", ignoring
Dec 28 16:14:22 CNCPi systemd[1]: cncjs.service: Service has no ExecStart=, ExecStop=, or SuccessAction=. Refusing.
Dec 28 16:21:57 CNCPi systemd[1]: /etc/systemd/system/cncjs.service:19: Executable "-port" not found in path "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin", ignoring
Dec 28 16:21:57 CNCPi systemd[1]: cncjs.service: Service has no ExecStart=, ExecStop=, or SuccessAction=. Refusing.
Dec 28 16:21:59 CNCPi systemd[1]: /etc/systemd/system/cncjs.service:19: Executable "-port" not found in path "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin", ignoring
Dec 28 16:21:59 CNCPi systemd[1]: cncjs.service: Service has no ExecStart=, ExecStop=, or SuccessAction=. Refusing.
  - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - 
  [i]  CNCjs is now started and can be accessed at: ( http://192.168.0.54 ) | ( http://192.168.0.54:8000 )   

  Setup IPtables   
  [✓]  Install Iptables  
  [✓]  Setup IPtables (allow access to port 8000 from port 80)  
  [✓]  Making Iptables Persistent  

  Finished   
  [i]  Script Finished, its recommended that you reboot your Raspberry Pi.
   └── To reboot run the command: sudo reboot   
pi@CNCPi:~ $ 
`

The Install CNCjs error points to a log, which I have to say is extremely long and I'm not sure what I'm looking for.

Thanks for any help.

AustinSaintAubin, 2020-12-31:
Looks like issue with service creation. Will run test with script. There might be issues with the latest Rasp Pi OS that was released this month.

What [Raspberry Pi OS](https://www.raspberrypi.org/software/operating-systems/) are you using?

AcrimoniousMirth, 2020-12-31:
@AustinSaintAubin I’m away from Pi atm so can’t give exact version but latest Raspbian at time of install (mid-Oct).

Is there a version of raspbian you’d recommend backdating to to try again?

aoturoa, 2021-01-04:
I had the same problem with Raspbian 2020-12-02 release. 
With the previous Raspbian version ([2020-08-20](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-08-24/2020-08-20-raspios-buster-armhf-lite.zip)) the installation runs smooth. 
Not sure if it mattered, but with the 2020-08-20 installation, I also unchecked any (opional) pendants during setup.

AustinSaintAubin, 2021-01-05:
Working to test and resolve this. Looks like issue with Rasp Pi OS 2020-12-02 release...

AcrimoniousMirth, 2021-01-05:
Will back-date to an earlier version of Raspbian and do a fresh install when I can (ideally tonight but maybe tomorrow). Will update with results.

AustinSaintAubin, 2021-01-12:
Working on IPtables fix... its somewhat related to first update of Pi.
I think your issue might be because you are using "Raspberry Pi OS with desktop and recommended software"
Try using "[Raspberry Pi OS with desktop](https://www.raspberrypi.org/software/operating-systems/)", not the full bloated version.

AustinSaintAubin, 2021-01-27:
Try with latest script, tested on Full and Standard.
Added condition to fix some issues with iptables.... but requires reboot.

TJ-Houghton, 2021-02-02:
I think I had the same problem.  I'm new to Raspberry Pi/Linux so I am struggling through this.  I tried installing CNCjs using the instructions on the main cncjs page but that didn't work.  I tried Austin's installation program and after running it 2 or three times tonight, it finally worked.  I have a Rii keyboard and am trying to install the cncjs keyboard pendant but don't understand the instructions.  It says to clone the repository to my work directory, but I am not sure what directory that is.  I think I know how to clone the repository but I just don't know where to put it.  I already installed the two libusb and lbudev packages.  Finally, it says to run bin/cncjs-pendant-keyboard.  Do I need to do that everytime I turn on my pi?  I don't plan on keeping it on because my SO3 is in my unheated garage and don't want my pi to freeze in the winter.

AustinSaintAubin, 2021-02-03:
Thanks for your input, post logs in the future some I can better debug. 
What version of Raspberry Pi and OS are you running?

I will work to automate adding the other addons to the Install Script for the Pi in the coming months.

Also don't worry too much about about board freezing, low temps ~32F are ok, Pi will operate at freezing.
I leave my on in the winter and do work. Its in an enclosure that keeps everything (controller & Pi) at a low, but even temperature in the winter.
https://www.raspberrypi.org/documentation/faqs/#pi-performance-temps

TJ-Houghton, 2021-02-03:
> Thanks for your input, post logs in the future some I can better debug.
> What version of Raspberry Pi and OS are you running?
>
> I will work to automate adding the other addons to the Install Script [...]

Austin,  Thanks for the response.  I did get the Shop Floor Table working, but since it doesn't have macros, I probably won't be using it.  I use macros for probing and doing M6 Tool Length Offsets.  I haven't been able to figure out the keyboard pendant, but that's taking a back burner right now.  My biggest issue is with Dropbox.  I do all my design and post-processing in my house and save the GCODE to a dropbox folder.  When I go out into the shop, I have a laptop that I can use cncjs and can access the GCode from my dropbox folder.  My plan was to replace the workshop laptop with the Pi, but I am going to have to figure out how to access the GCODE files because Dropbox doesn't show up in the GUI when loading gcode files in cncjs (or any other GUI).  I don't think that is a cncjs issue, but a Dropbox/PI issue with interfacing with the GUIs.  I just need to put a little more thought into it and may have an extra step where I copy the .nc files to the PI before I head out to the workshop.

Thanks again,
TJ

AustinSaintAubin, 2021-02-03:
I will look into adding some notes on dropbox integration with the watch folder.
In the meantime, checkout: https://github.com/andreafabrizi/Dropbox-Uploader

TJ-Houghton, 2021-02-03:
Thanks.  I had that downloaded and it seemed to be working...until tonight.  So, I decided to reflash my SD card and will reinstall everything.  :(

AustinSaintAubin, 2021-03-24:
Odd, are you still having issues?

TJ-Houghton, 2021-03-24:
@AustinSaintAubin   I decided to just use the Dropbox website to download files.  It was much easier than trying to figure out CLI instructions.  Once I reflashed my SD card CNCjs was working fine, but I stopped using CNCjs right now, though, because of the inability to handle files greater than about 5MB.  There are not a lot of options but I haven't done a lot of experimenting with any right now - too many other projects around the house that I need to get done, and that don't require G code.

AustinSaintAubin, 2021-03-25:
Odd that you are having issues with 5MB files.
I have been able to handle files as large ~25MB on my RPi 3B when doing laser engraving.

TJ-Houghton, 2021-03-25:
@AustinSaintAubin I posted on other forums and the general consensus is that somewhere are ~5-6 MB files is where the CNCjs server shuts down.  I have successfully run files smaller than that, but occasionally I get a message that the server shut down but the SO3 continues to operate.  Bigger files just won't load and I can't send them to my SO3.

One other bug that I have been having is that my RPi 4 won't save my widget settings.  Every time I reboot my Pi, the default widgets show up and I need to go in and hide the ones I don't want.  Fortunately my macros are saved.

asagayakita, 2021-04-18:
It didn't work with Raspberry Pi OS 64-bit.

comqdhb, 2021-06-23:
I have a pi zero w
The point of failure was the lack of official node.  Now have to use unofficial version of node for armv6l

installed node outside of the script and then ran script un-ticking need to install node

```
mkdir node
cd node
wget https://unofficial-builds.nodejs.org/download/release/v10.24.1/node-v10.24.1-linux-armv6l.tar.xz
tar xvfJ node-v10.24.1-linux-armv6l.tar.xz 
sudo cp  -R node-v10.24.1-linux-armv6l/* /usr/local
cd  ..
rm -rf node
```

TJ-Houghton, 2021-07-08:
@comqdhb  I finally found time to try your suggestion.  I ran your script and it seemed to work fine, however, when I ran Austin's script, it didn't give me the option to skip the node install, so I don't know if his script overwrote the armv61 version of node.  The CLE screen said that, during Austin's script, nodejb v10.24.1 was installed but it was unclear if it was the armv61 variant or whatever gets installed in Austin's script.  

In any event, I tried loading a large (33MB) file and it choked on it - shutting down the server with an error message.  The cncjs GUI eventually finished uploading the file and showed a visual representation in the viewer, but the server had already shut down so I couldn't send it to my Shapeoko.  

Is it possible to restart the server after it finished loading and then run it?  Or is too late at that point?

I'll try reinstalling the armv61 variant of nodejs (without uninstalling cncjs) and see if that fixes the problem

AustinSaintAubin, 2021-07-10:
@TJ-Houghton, you have the large gcode file to test. Will test if you post it.
Script is designed to install on a fresh install of Raspberry Pi OS. 
Will test this weekend and see if any recent changes in public repos are causing issues.

TJ-Houghton, 2021-07-10:
Austin,

Thanks.  Attached is a 33MB Gcode file that was trying to load.

[Setup 3G(Finish) Front (TOP).zip](https://github.com/cncjs/cncjs-pi-raspbian/files/6795459/Setup.3G.Finish.Front.TOP.zip)

comqdhb, 2021-07-12:
> @comqdhb I finally found time to try your suggestion. I ran your script and it seemed to work fine, however, when I ran Austin's script, it didn't give me the option to skip the node install, so I [...]

[image]

TJ-Houghton, 2021-07-12:
I don't recall if that screen came up last time I installed it.  I am re-installing it now on a fresh SD card

TJ-Houghton, 2021-07-12:
I finished installing armv61 and cncjs (without node.js selected) and still getting the same error.  Within seconds of trying to load a large file, I get an error message saying the server shutdown and I can't load my file

AustinSaintAubin, 2021-08-06:
Major improvements to the install script, try now.

TJ-Houghton, 2021-08-09:
@AustinSaintAubin , I gave it a try and no difference.  I tried loading a 10MB bile last night and almost immediately recevied a notice that the server shut down and then cncjs got stuck with a spinning wheel trying to load the file.  I was able to get carbide motion installed on my RPi, so it's not as big of a problem now, but CM doesn't support all the features that cncjs does, so it's a little bit of a hassle (but at least I can run my large files :) )

Fortunately, I don't do a lot of work with large files.

AustinSaintAubin, 2021-08-11:
You able to share that file with me... might be something in the g-code.
I just tested with 10.8MB file from here: https://easel.inventables.com/projects/BBROM2ec2CNsAwHj9mDj7g
Then ran though here: https://github.com/AustinSaintAubin/gcode-depths-to-shades

Rendered just fine.
[image]

[image]

TJ-Houghton, 2021-08-11:
@AustinSaintAubin , thank you for trying to troubleshoot this with me.  Unfortunately, I am still not having any luck.  the 10MB file I tried loading was actually a zip file of the 30MB file I sent you earlier (oops).  I have been able to successfully upload a 4 MB file to my cncjs, but haven't been able to load anything any bigger.  I downloaded your 10MB tiger tattoo, and am still waiting for it to load.  It has NOT told me that the server shut down, so that's good news, but it hasn't gotten to the rendering stage either, and it has been loading for a while (at least a half hour)

beanbagbonanza, 2022-02-14:
@comqdhb This method is working for my board on the most current raspbian OS:
  [i]  Detected HW: Raspberry Pi 3 Model A Plus Rev 1.0   
  [i]  Detected OS: [ raspbian | 11 | No GUI ]  

Thanks. This was a pain until I ran across this page.

beanbagbonanza, 2022-03-13:
@AustinSaintAubin This @comqdhb fix, or some version of it, would be great set up as an option in the original install script for those of us experiencing this issue. Thanks to you both. I've done the manual install for the last time (I hope).  
Raspberry Pi 3 A
raspbian | 11 | No GUI

AustinSaintAubin, 2024-07-11:
Updated the [cncjs-pi-raspbian](https://github.com/cncjs/cncjs-pi-raspbian) install script.
The [Lite x64](https://downloads.raspberrypi.org/raspios_lite_arm64_latest) edition is now recommended.
Recommended RP4 or greater hardware. 

Works with:
- NodeJS: v18.19.0
- NPM: v9.2.0

closed 2024-07-11 by AustinSaintAubin