cncjs/cncjs-pi-raspbian #18: web interface problems

timurbautdinov, 2021-03-22:
Hi. I've installed the cncjs on my raspberry pi zero according [this](https://github.com/cncjs/cncjs-pi-raspbian) instruction.
Everything run well, but after rebooting I could't see the web interface. 
Pi is running and connected to local host, the adress is true, what I do wrong? Thank you!

AustinSaintAubin, 2021-03-24:
Looks like crontab did not run... odd

Reboot and Run this:
```
sudo iptables -t nat -I PREROUTING -p tcp --dport 80 -j REDIRECT --to-port 8000; sudo netfilter-persistent save; sudo netfilter-persistent reload; sudo bash -c "iptables-save > /etc/iptables/rules.v4"; sudo bash -c "ip6tables-save > /etc/iptables/rules.v6"
```

timurbautdinov, 2021-03-26:
One more time reinstalled, but nothing happens( 
Will I try this [instruction](URL="https://raw.githubusercontent.com/cncjs/cncjs-pi-raspbian/master/cncjs_install.sh"
curl -sSL ${URL} | bash)? Or maybe you could offer me the easiest way? Thanks for your support!

AustinSaintAubin, 2021-04-02:
@timurbautdinov the script should be the easiest way.
Post the syslog after running the install `tail -f -n 50 /var/log/syslog`

AustinSaintAubin, 2021-08-06:
Updated the script, does not need iptables anymore.
Do with fresh install.