cncjs/cncjs-pi-raspbian #19: CNCjs installed but wont start

raptorfa22, 2021-08-22:
Description
I was able to install CNCjs on my raspberry pi 4 and upon reboot I get the popup screen stating that the server is starting and with the directions on how to exit to chrome kiosk and to exit the UI. After clicking ok on this popup nothing happens. I have even tried opening from a web browser on another computer. I will be using CNCjs to control a Genmitsu 3018 pro with the y axis extension to a 3040 setup.

Micro SD card used:
SanDisk 128GB Extreme microSDXC UHS-I Memory Card with Adapter - Up to 160MB/s, C10, U3, V30, 4K, A2, Micro SD - SDSQXA1-128G-GN6MA

Versions
CNCjs: 1.9.x
Node.js: 6.x
NPM: 5.x
How Do You Install CNCjs?
Installed using the following:
URL="https://raw.githubusercontent.com/cncjs/cncjs-pi-raspbian/master/cncjs_install.sh"
curl -sSL ${URL} | bash

A00 through A07 were selected during installation from the splash screen
All pendant and widget options in the splash screen were selected

CNC Controller
 Grbl

raptorfa22, 2021-08-22:
I was able to get CNCjs to start by typing cncjs in the command prompt in RPI OS but i still cannot remotely access using the local host command or ip address. Is there a way to set a static IP address instead of the ip address changing every time I restart the RPI?

AustinSaintAubin, 2021-08-28:
Awesome! Glade it was able to be figured out. Thanks for sticking with it.

closed 2021-08-28 by AustinSaintAubin