cncjs/cncjs-pi-raspbian #22: Cant Connect

MichaelJmad, 2021-11-27:
Run the setup as it is posted and after a reboot, nothing cant connect it gives an ERR_CONNECTION_REFUSED
Tried to run the CNC command npm etc says command not found please help, 

P.S. thanks for the scripts

DAV3103, 2021-12-03:
Yup I am having the same problem seems to be NPM

MichaelJmad, 2021-12-04:
I have been able to get it working some but it's very slow/ not responds...
Note I had to install using the CNCJS for Pi guide and not the script. GIT was having issues the day I was installing so that might have been it.

AustinSaintAubin, 2024-07-11:
Updated the [cncjs-pi-raspbian](https://github.com/cncjs/cncjs-pi-raspbian) install script.
The [Lite x64](https://downloads.raspberrypi.org/raspios_lite_arm64_latest) edition is now recommended.
Recommended RP4 or greater hardware. 

Works with:
- NodeJS: v18.19.0
- NPM: v9.2.0

closed 2024-07-11 by AustinSaintAubin