cncjs/cncjs-pi-raspbian #23: Can't install on Raspberry Pi Zero W Rev 1.1

alexcaglairi, 2022-02-01:
Hi,
I have the last  Raspberry Pi Zero W Rev 1.1.
The steps are:

1. Using Raspberry Pi Imager v 1.6.2 i choose "Raspberry Pi OS Lite 32 bit"
2. Inserting Micro SD into Raspi
3. Login via SSH
4. Run "sudo apt update"
5. Run "URL="https://raw.githubusercontent.com/cncjs/cncjs-pi-raspbian/master/cncjs_install.sh" curl -sSL ${URL} | bash"

[image]

[image]

[image]

**First Error**
[image]

**.. but solved**
[image]

**The problem is this:**
[image]

At this point i tried to manually install npm using "sudo apt install npm -y".
Now typing "npm -v" i have "7.5.2" output
After i relaunch cncjs installer running only the cnc js installation with this output:
[image]

### System Info
```
uname -a
	Linux laser 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux
	
cat /etc/*-release
	PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
	NAME="Raspbian GNU/Linux"
	VERSION_ID="11"
	VERSION="11 (bullseye)"
	VERSION_CODENAME=bullseye
	ID=raspbian
	ID_LIKE=debian
```

AustinSaintAubin, 2022-02-02:
Thanks for the detailed info.
Just an FYI, I generally don't recommend running CNCjs on the Pi Zero, because of performance issues, but let's take a look.

Looks like the Pi Zero requires a different node binary. Which you installed manually (good).

Looking at your latest output results it looks like CNCjs should be working... i am not sure I understand your current issue, aside for script not handling CNCjs on Raspberry Pi Zero.

AustinSaintAubin, 2022-02-02:
Might use a Raspberry Pi Zero 2, it uses the same processor architecture as the newer Pis, like the Pi4.

DavidJJ, 2022-11-23:
Did anyone ever solve this issue?  I'm running into the same problems trying to build a cncjs interface to my TimSav needle cutter.

AustinSaintAubin, 2024-07-11:
Updated the [cncjs-pi-raspbian](https://github.com/cncjs/cncjs-pi-raspbian) install script.
The [Lite x64](https://downloads.raspberrypi.org/raspios_lite_arm64_latest) edition is now recommended.
Recommended RP4 or greater hardware. 

Works with:
- NodeJS: v18.19.0
- NPM: v9.2.0

closed 2024-07-11 by AustinSaintAubin