cncjs/cncjs-pi-raspbian #24: How can I make the mouse cursor visible again?

plangenh, 2022-03-03:
How can I show the mouse cursor again? 
I have cncjs running on the pi with a screen and a keyboard but theres no mouse cursor available. 
Already looked into script but couldn't find anything related to the cursor. 
thanks a lot in advance!

AustinSaintAubin, 2022-03-04:
Remove the '-nocursor' flag below

# Start X automatically on boot
msg i "Setting StartX to start UI on login"
cat >> "${HOME}/.bashrc" << "EOF"
# Start Web Kiosk
[[ -z $DISPLAY && $XDG_VTNR -eq 1 ]] && startx -- -nocursor
EOF

plangenh, 2022-03-06:
> Remove the `-nocursor` flag below
> 
> ```
> # Start X automatically on boot
> msg i "Setting StartX to start UI on login"
> cat >> "${HOME}/.bashrc" << "EOF"
> # Start Web Kiosk
> [[ -z $DISPLAY && $XDG_VTNR -eq 1 ]] && startx -- -nocursor
> EOF
> ```

removing the `-nocursor`  flag in the script on line 946 before a clean install on a fresh system results in a black screen with a white cursor after the `Waiting for CNCjs server to start` dialogue box disappeared.

AustinSaintAubin, 2022-05-30:
Sorry for the late reply.
Are you using Rasp OS Desktop or Light?
Edit the kiosk script starting parameters with `nano ~/.bashrc`.

closed 2024-07-11 by AustinSaintAubin