cncjs/cncjs-pi-raspbian #25: Error 1 using method 1 or 2 :  NPM is not installed.

henridbr, 2022-05-26:
Hi,

I get this error 1 using method 1 or 2. 
NPM is not installed. 

`  =====================================================
  [i]  Detected HW: Raspberry Pi 3 Model B Rev 1.2
  [i]  Detected OS: [ raspbian | 11 | No GUI ]
   └── Raspberry Pi OS | Light | 2022-04-04
  [✓]  Detected OS is compatable with this install script.
  [✓]  Updating System Packages
  [✓]  Upgrading System Packages (this can take a while, please wait)
  [✓]  Upgrading System Distribution (this can take a while, please wait)
  [✓]  Fixing Broken Packages (if any)

  Setup Node.js & NPM via Package Manager
  [✓]  Removing any Old NodeJS or NPM Packages
  [✓]  Removing Un-needed Packages
  [✓]  Installing Node.js v10.x Package Source
  [✓]  Installing Node.js v10.x via Package Manager
  [✓]  Installing Build Essential
  [i]  NodeJS:  v12.22.5        |        /usr/bin/node
  [i]  NPM:     Not Installed

  Install CNCjs
cncjs_install.sh: line 565: npm: command not found
cncjs_install.sh: line 574: CNCJS_VERSIONS: bad array subscript
cncjs_install.sh: line 576: CNCJS_VERSIONS: bad array subscript
  [✗]  Installing CNCjs (v) with NPM | Error Code: 1
   └── Try to re-run this part of the script after rebooting.
  Latest Syslog Entries - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
May 26 23:18:32 raspberrypi cncjs_install.sh: Installing Build Essential
May 26 23:18:36 raspberrypi cncjs_install.sh: NodeJS:   v12.22.5        |        /usr/bin/node
May 26 23:18:36 raspberrypi cncjs_install.sh: NPM:      Not Installed
May 26 23:18:36 raspberrypi cncjs_install.sh: Install CNCjs
May 26 23:18:44 raspberrypi cncjs_install.sh: Installing CNCjs (v) with NPM
May 26 23:18:44 raspberrypi cncjs_install.sh: sudo: npm: command not found
  - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
  [✓]  Set User TTY Permissions

  Download & Install CNCjs Pendants & Widgets   [ /root/.cncjs ]
=============================================`

Any idea ?

Henri

AustinSaintAubin, 2022-05-27:
Looks like NPM is not installing...

run, and share output.
`npm -v`

run
`npm install -g npm@latest`

AustinSaintAubin, 2022-05-27:
Might try older raspberry pi image.... i need to test the latest.

henridbr, 2022-05-27:
Hi,

pi@raspberrypi:~ $ npm -v
-bash: npm: command not found
pi@raspberrypi:~ $ node -v
v12.22.5
pi@raspberrypi:~ $ npm install -g npm@latest
-bash: npm: command not found

I had hard time trying to install CNCjs using different methods exposed but failed on errors some about scripts missing I don't know why. So I thank you for your script which seems to save days of headhaches to many despite this issue. 

I'll try an older  raspberry pi image.

Henri

henridbr, 2022-05-27:
Hi,

The first older working version is [raspios_lite_armhf-2021-05-28](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/)
with  node v10.24.1 and npm v6.14.12
==========================================
  [i]  Detected HW: Raspberry Pi 3 Model B Rev 1.2
  [i]  Detected OS: [ raspbian | 10 | No GUI ]
   └── Raspberry Pi OS | Light | 2021-05-07
  [✓]  Detected OS is compatable with this install script.
==========================================
Maintaining such a script with Raspberry.org publishing 3 new versions per year is a considerable work to do alone.
I understand that. Perhaps could you indicate the last OS working version ?
Thank you, it really helps me.
Henri

henridbr, 2022-05-27:
Hi,

The first older working version is [raspios_lite_armhf-2021-05-28](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/)
****************************************************
  [i]  Detected HW: Raspberry Pi 3 Model B Rev 1.2
  [i]  Detected OS: [ raspbian | 10 | No GUI ]
   └── Raspberry Pi OS | Light | 2021-05-07
  [✓]  Detected OS is compatable with this install script.
*****************************************************************
Maintaining such a script with Raspberry.org publishing 3 new versions per year is a considerable work to do alone.
I understand that. Perhaps could you indicate the last OS working version ?

Thank you, it really helps me.
Henri

AustinSaintAubin, 2022-05-27:
Thanks for testing and posting back. Looks like some breaking changes with latest raspberry pi image release.

Note to self. 
- Add latest tester raspberry pi image information to the read-me.
- Update script to work with latest raspberry pi image.

closed 2022-05-30 by AustinSaintAubin