cncjs/cncjs-pi-raspbian #26: Webbrowser does not open server page after script installation

JupaCreations, 2022-05-30:
I get lost. 🙁
2 years ago I installed cncjs on a pi as server. Do not know anymore how I did it... 
After two years I fire up the system to dry test again.
By webbrowser all seems to start up and work fine.
Then a server connection error pops up.
Reboot Pi but after a while same thing.
Refresh browser but same problem.
Take a fresh SD card but putting cncjs on it with the instrcutions from the faceboook group and it failed several times.
In the end of every installation the webbrowser does not connect to the server address.
So i formated the working one and that now also will not install cncjs fully. 
SD cards formated quick and fully does not matter to the outcome. 
Linux is not my native language so pulling some hair here to get past the errors to have cncjs working again.

PI 3B with LAN cable connected.
Used 2 different PI's to troubleshoot to exclude hardware.
Sandisk 16 and 32 gb cards.
Then I was notified of this script so I formatted an SD card and installed Lite Pi OS. 
Ran the code for the script.
It finishes the script but for what I understand did not install node.js and NPM.
So I ran
sudo apt install -y nodejs build-essential npm
and
sudo npm install -g npm@latest
one after each other with a sudo reboot inbetween.
Whe I log in with the webbrowser on 192.168.1.96 with or without port :80 or :8000 added it does not connect. The PI has connection otherwise to the private network as I can log in with Putty.
I tried above two times to be sure I did not missed anything on my side.

What can cause the CNCJS server not to be found?
Did I missed something?
Any guidance to solve this mistery would be appriciated.

Paco

[image]

AustinSaintAubin, 2022-05-30:
Try again with latest

JupaCreations, 2022-06-01:
Hello Austin,
Thanks, I will do now. 1-6-2022 08.15 CET
BTW is there an option after the install if the connection with browser does not work to check if the server is running correctly?
I saw cncjs at the commandline should give an output according to an other installation guide.
But as you can see from the screenshot is says command not found.

JupaCreations, 2022-06-01:
OK, this is my route for trouble shooting.
No SD card format.
I ran the script with the option to remove the NODEJS and NPM which already exist.
The script finishes and removed the NODEJS and NPM and installed new ones.
Install of CNCJS lates version failed as is mentioned in the script output.
Browser can not be reached.
I have no screenoutput of the script. (sorry)
So next try.
No SD card format.
Start script and do not change any option in the startup screens.
Script finishes but again reports that CNCJS install failed,
I have add the screenshot of the script output.

Paco
[image]

JupaCreations, 2022-06-02:
Browser problem solved.

I tried installation option 1 one more time with same result.
I formatted the SD card put new image on but then choose for install option 2.
Just my luck? The script ran without an error on the 9.25 cncjs install.
After a reboot the browser showed the page.

Paco

AustinSaintAubin, 2022-06-05:
@JupaCreations, that is odd... looks like the CNCjs install directory had content in it... that could have halted the install.
Glad its working with fresh install.

closed 2022-06-05 by AustinSaintAubin