cncjs/cncjs-pi-raspbian #27: Im not sure why it wont work

Todderbod, 2022-08-06:
_______   ________  _
    / ____/ | / / ____/ (_)____
   / /   /  |/ / /     / / ___/
  / /___/ /|  / /___  / (__  )
  \____/_/ |_/\____/_/ /____/
                  /___/

  Installing CNCjs
   - https://cnc.js.org
   - https://github.com/cncjs

  CNCjs is a full-featured web-based interface for CNC controllers running Grbl,                                                                                                                                                              Marlin, Smoothieware, or TinyG.
  For a more complete introduction, see the Introduction section of the wiki pag                                                                                                                                                             e ( https://github.com/cncjs/cncjs/wiki/Introduction ).
  NOTE: This installer logs to syslog. You can view the syslog, with terminal co                                                                                                                                                             mmand: tail -f -n 50 /var/log/syslog
  ==========================================================================
  [i]  Detected HW: Raspberry Pi 3 Model B Plus Rev 1.3
  [i]  Detected OS: [ raspbian | 10 | Compatible GUI ]
   └── Raspberry Pi OS with desktop and recommended software | Full | 2021-05-07
  [✓]  Detected OS is NOT compatable with this install script!
  [i]  This installer is designed for the [Raspberry Pi](https://www.raspberrypi.org)

what do I need to do Im new to this

AustinSaintAubin, 2022-08-06:
It seems like the setup wizard might not be running... or some of the output is missing... can you try rerunning the script and repost the output?

Todderbod, 2022-08-08:
Sorry I did not get right back. When I run the script it goes to a blue screen that says cncjs install options i select ok the next screen asks me about pendants and widgets which I have no clue what to select but when I hit ok it goes back to the black screen were it shows what I uploaded before. Im going to start all over from reinstalling raspian and see what happens

AustinSaintAubin, 2024-07-11:
Updated the [cncjs-pi-raspbian](https://github.com/cncjs/cncjs-pi-raspbian) install script.
The [Lite x64](https://downloads.raspberrypi.org/raspios_lite_arm64_latest) edition is now recommended.
Recommended RP4 or greater hardware. 

Works with:
- NodeJS: v18.19.0
- NPM: v9.2.0

closed 2024-07-11 by AustinSaintAubin