cncjs/cncjs-pi-raspbian #28: Fresh Install no web interface connection

Thesourceminer, 2022-11-07:
Used Rasp Pi Imager v1.7.3
Installed Pi OS Lite (32 bit) Released 2022-09-22
Raspberry Po 3 Model A Plus
Upon connection to SSH (Preconfigured WIFI and SSH during imager. )
Ran the Install Script for CNCjs 
- URL="https://raw.githubusercontent.com/cncjs/cncjs-pi-raspbian/master/cncjs_install.sh"
- curl -sSL ${URL} | bash
Went through the wizard Selected: A00-A05 (Even later tried A12)
Selected Pendant Shopfloor Tablet.
Selected version 1.9.27
Script runs through all the sequences without any error. 
States to connect to the address I selected which was port 80
Rebooted
When attempting to run cnc or cncjs at the prompt, I get the command not found. 

Looked into the home/user/.cncjs folder and found no cncrc.cfg "ExecStart= --port 80 --config "/home/pi/.cncjs/cncrc.cfg""

Status of Service - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    ● cncjs.service - CNCjs is a full-featured web-based interface for CNC controllers running Grbl, Marlin, Smoothieware, or TinyG.
     Loaded: loaded (/etc/systemd/system/cncjs.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Sun 2022-11-06 19:55:06 PST; 2s ago
       Docs: https://github.com/cncjs/cncjs
   Main PID: 8543 (code=exited, status=0/SUCCESS)
        CPU: 54ms

 [i]  CNCjs is now started and can be accessed at: ( http://localhost:80 ) | ( http://192.168.4.68:80 )

binarylefmart, 2023-01-14:
Hi ! any news about this issue ? have the same here.

AustinSaintAubin, 2023-01-14:
I think might be related to [#29], though I have not been able to recreate the issue on a Raspberry Pi 4.
Trying to make time to add options allowing user to select which NodeJS & NPM version they would like to install.

AustinSaintAubin, 2024-07-11:
Updated the [cncjs-pi-raspbian](https://github.com/cncjs/cncjs-pi-raspbian) install script.
The [Lite x64](https://downloads.raspberrypi.org/raspios_lite_arm64_latest) edition is now recommended.
Recommended RP4 or greater hardware. 

Works with:
- NodeJS: v18.19.0
- NPM: v9.2.0

closed 2024-07-11 by AustinSaintAubin