cncjs/cncjs-pi-raspbian #29: Install on RPI 3 B Plus causing issues with JS memory heap

thatSFguy, 2023-01-08:
The current NodeJS 12 version is causing issues with the CNCJS installation throwing a JS memory heap / running out of memory error.   (Sorry didn't copy the exact error)

After following the instructions several times found at https://cnc.js.org/docs/rpi-setup-guide/ , I found this install script.  However, it had the same issue. Finally I found the issue opened with the RPI Zero, which seems to have a similar NodeJS problem.

The solution presented there also worked for me by first installing Nodejs 10 with:

`mkdir node
cd node
wget https://unofficial-builds.nodejs.org/download/release/v10.24.1/node-v10.24.1-linux-armv6l.tar.xz
tar xvfJ node-v10.24.1-linux-armv6l.tar.xz 
sudo cp  -R node-v10.24.1-linux-armv6l/* /usr/local
cd  ..
rm -rf node`

Then run the rest of the script as described on the main page, but deselect the Node install/upgrade option :

`URL="https://raw.githubusercontent.com/cncjs/cncjs-pi-raspbian/master/cncjs_install.sh"
curl -sSL ${URL} | bash`

[image]

AustinSaintAubin, 2023-01-09:
Just to confirm, NodeJS 12 is causing you issues and crashing?
I have not encountered this issue. Will try to do some more testing if time permits.
Thanks for posting a work around here in the meantime.

thatSFguy, 2023-01-09:
Correct.  I tried NodeJS 12, 16,18, and 19.  all ended up with a memory heap crash.

thatSFguy, 2023-01-09:
BTW, Thanks so much for putting together this amazing piece of software!  I've used Octoprint for several years and was looking for something similar in CNC.  Although I've not made my first cuts yet with my modded 1998 Denford micromill, I think CNCJS will really ease the transition.  Thanks again!

AustinSaintAubin, 2023-01-09:
@thatSFguy, I can only take credit for some of the raspberry pi integration work. @cheton is the original author, and many others have contributed. 

Will try to conduct some testing with an RP3B... but I don't one that is not in use at the moment... when RP4 become available I recommend upgrading to one. 

Glad you are enjoying the software, and the raspberry pi install script made some of the work simpler.

thatSFguy, 2023-01-10:
I just wanted to point out that when the CNCJS install fails the install script still continues on, even if it can't find the cncjs executable for further setup.  It maybe good to have a check to ensure the install was successful, prior to continuing the script.

P.S. I also tried export NODE_OPTIONS="--max-old-space-size=8192" to no avail.

Cjkeenan, 2023-01-14:
I am having this same issue with my RPi 2B, I modified your script to install nodejs 10 with "Raspbian" 11 since nodejs 10 seems to work for a lot of people, so I will let you know if that helps at all.

AustinSaintAubin, 2023-02-10:
I think this might just be an issue when installing on `Raspberry Pi OS with desktop`.
I have been testing on `Raspberry Pi OS Lite` with no issues. 

I think `Raspberry Pi OS with desktop` comes pre-installed with a version of [Node.js](https://nodejs.org/)....
Still doing some testing to figure out what is the cause.
Might try uninstalling Node.js first, then re-running the script and installing everything...

--------

@Cjkeenan CNCjs is still one of the best options to spite the bugs (in my opinion). 
Though is heavy and best to run on RP3 or RP4 (or similar), with a fast SD card or flash drive.

It does take a lot of effort to build a project like this.
If you like it support it by [becoming a backer](https://github.com/cncjs/cncjs#backers). 
_I personally do not receive any of these contributions, I just want to see the project grow._

Cjkeenan, 2023-02-10:
> Though is heavy and best to run on RP3 or RP4 (or similar), with a fast SD card or flash drive.

I have been running this without many issues on my 2B. How would I identify that I am running into performance issues? The webpage seems plenty snappy. Would I be seeing a difference in cut quality like you sometimes do when using Octoprint on a less powerful system?

AustinSaintAubin, 2023-02-11:
@Cjkeenan, run `htop` while CNCjs is preforming operations.
Video stream is also a heavy load on the CPU.

Cjkeenan, 2023-02-11:
I'm not doing any video streaming

thatSFguy, 2023-02-11:
Fyi-  I did not install the desktop version when I got the error on my 3B.  Hope that helps.

closed 2024-07-11 by AustinSaintAubin