cncjs/cncjs-pi-raspbian #30: Where is the .cncrc located?

deependdesigns, 2023-01-22:
I am trying to install https://billiam.github.io/cncjs-pendant-streamdeck/docs/installation/ and am having trouble locating the .cncrc location.   Thank you.

deependdesigns, 2023-01-22:
Nevermind about that. I went ahead and read through the script and found that it creates a service:

/etc/systemd/system/cncjs.service

It turns out that I too am having the same issues as https://github.com/cncjs/cncjs-pi-raspbian/issues/29 on a Raspberry Pi 3B+.  I was able to get everything up and running with NodeJS V10.24.1 however when I sent to setup https://github.com/Billiam/cncjs-pendant-streamdeck I ran into a heap of errors around my version of Node. It seems V10.24.1 is the only version working on the 3B+ at the moment.

What would it take to get the Stream Deck pendant incorporated within your project and working on a Raspberry Pi 3B+? I'd be more than happy to help and test, maybe donate?

AustinSaintAubin, 2023-01-26:
@deependdesigns, glad you got CNCjs running on your Raspberry Pi 3B+. I am having issues locating a Pi 3B to test on to update the script with the notice.

AustinSaintAubin, 2023-01-26:
@deependdesigns, you might have a look at the following to write a Stream Deck pendant control widget.
- https://github.com/cncjs/cncjs-controller
- https://github.com/cncjs/cncjs-pendant-keyboard
- https://github.com/cncjs/cncjs-pendant-ps3

deependdesigns, 2023-01-26:
So there is a JS memory heap error when installing with other versions I tried other than 10.24.1 but I found that I can install using that version of node and then use n to upgrade to 14.x after installing cncjs 1.10.0.

By doing this I was able to get the streamdeck pendant running. I will say, if you haven't tried the streamdeck pendant....do it! I now don't need anything in the garage and it's very intuitive.  You just need a basic understanding of json to customize...and it's exceptionally customizable.

deependdesigns, 2023-01-26:
That being said. I am more than happy to help you test and get the script working on the 3b

AustinSaintAubin, 2024-07-11:
Updated the [cncjs-pi-raspbian](https://github.com/cncjs/cncjs-pi-raspbian) install script.
The [Lite x64](https://downloads.raspberrypi.org/raspios_lite_arm64_latest) edition is now recommended.
Recommended RP4 or greater hardware. 

Works with:
- NodeJS: v18.19.0
- NPM: v9.2.0

closed 2024-07-11 by AustinSaintAubin