cncjs/cncjs-pi-raspbian #31: Issues installing 1.10.1 on RPi 2B Rev 1.1

Cjkeenan, 2023-03-24:
Anyone have any luck upgrading 1.10.1? I have rereun the script twice (first time with basically default options, and second with the delete old NPM packages and NodeJS) but I am running into the same error each time (after reboot):

Any ideas would be greatly appreciated. I assume it should work because 1.9.27 is running currently and nodejs 12 is supported in 1.10.1 afaik.

```
pi@rpi2b-cncjs:~ $ curl -sSL ${URL} | bash
CNCjs Installer v1.4.4(2023-02-10) by: Austin St. Aubin
     _______   ________  _
    / ____/ | / / ____/ (_)____
   / /   /  |/ / /     / / ___/
  / /___/ /|  / /___  / (__  )
  \____/_/ |_/\____/_/ /____/
                  /___/

  Installing CNCjs
   - https://cnc.js.org
   - https://github.com/cncjs

  CNCjs is a full-featured web-based interface for CNC controllers running Grbl, Marlin, Smo
  For a more complete introduction, see the Introduction section of the wiki page ( https://
  NOTE: This installer logs to syslog. You can view the syslog, with terminal command: tail
  ==========================================================================
  [i]  Detected HW: Raspberry Pi 2 Model B Rev 1.1
  [i]  Detected OS: [  [✓] raspbian  |  [✓] 11  |  [!] No GUI (or) Incompatable GUI  ]
   └── Raspberry Pi OS | Light | 2022-09-22
  [✓]  Detected OS is compatable with this install script.
  [✓]  Updating System Packages
  [✓]  Upgrading System Packages (this can take a while, please wait)
  [✓]  Upgrading System Distribution (this can take a while, please wait)
  [✓]  Fixing Broken Packages (if any)

  Setup Node.js & NPM via Package Manager
  [✓]  Removing any Old NodeJS or NPM Packages
  [✓]  Removing Un-needed Packages
  [✓]  Installing Node.js & NPM via Package Manager
  [✓]  Installing Build Essential
  [i]  NodeJS:  v12.22.12       |        /usr/bin/node
  [i]  NPM:     v7.5.2  |        /usr/bin/npm

  Install CNCjs
  [✗]  Installing CNCjs (v1.10.1) with NPM | Error Code: 1
   └── Try to re-run this part of the script after rebooting.
  Latest Syslog Entries - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    Mar 24 15:33:14 rpi2b-cncjs bash: npm ERR! code FETCH_ERROR
Mar 24 15:33:14 rpi2b-cncjs bash: npm ERR! errno FETCH_ERROR
Mar 24 15:33:14 rpi2b-cncjs bash: npm ERR! invalid json response body at https://registry.npmjs.org/react reason: Unexpected end of JSON input
Mar 24 15:33:15 rpi2b-cncjs bash:
Mar 24 15:33:15 rpi2b-cncjs bash: npm ERR! A complete log of this run can be found in:
Mar 24 15:33:15 rpi2b-cncjs bash: npm ERR!     /root/.npm/_logs/2023-03-24T22_33_14_758Z-debug.log
  - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
  [✓]  Set User TTY Permissions
  [⣾]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch  ( /home/pi/.cncjs )   [⣽]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch  ( /home/pi/.cncjs )   [⣻]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch  ( /home/pi/.cncjs )   [⢿]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch  ( /home/pi/.cncjs )   [⡿]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch  ( /home/pi/.cncjs )   [⣟]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch  ( /home/pi/.cncjs )   [⣯]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch  ( /home/pi/.cncjs )   [⣷]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch  ( /home/pi/.cncjs )   [✓]  Creating CNCjs Directory for Addons / Extentions / Logs / Watch  ( /home/pi/.cncjs ) 

  Create CNCjs Service for Autostart
  [i]  Creating CNCjs Service w/ Systemd
  [✓]  Editing CNCjs Service Start Port
  [i]  Creating CNCjs Service Settings File (Optional)
  [✓]  Editing CNCjs Service Start Options: OPTIONS HERE
  Service Settings: /etc/systemd/system/cncjs.service - - - - - - - - - - - - - -
    ExecStart=/usr/local/bin/cncjs --port 80 --config "/home/pi/.cncjs/cncrc.cfg" --watch-directory "/home/pi/.cncjs/watch"

    Note: These settings can be changed with command: sudo systemctl edit --full cncjs
      Check CNCjs Server Status with: sudo service cncjs status
  - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
  [✓]  Reloading Service Deamon
  [✓]  Starting Service
  [✓]  Enabling Service
  Status of Service - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
    ● cncjs.service - CNCjs is a full-featured web-based interface for CNC controllers running Grbl, Marlin, Smoothieware, or TinyG.
     Loaded: loaded (/etc/systemd/system/cncjs.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2023-03-24 15:33:55 PDT; 3s ago
       Docs: https://github.com/cncjs/cncjs
   Main PID: 22522 (node)
      Tasks: 7 (limit: 1595)
        CPU: 3.947s
     CGroup: /system.slice/cncjs.service
             └─22522 node /usr/local/bin/cncjs --port 80 --config /home/pi/.cncjs/cncrc.cfg --watch-directory /home/pi/.cncjs/watch

Mar 24 15:33:55 rpi2b-cncjs systemd[1]: Started CNCjs is a full-featured web-based interface for CNC controllers running Grbl, Marlin, Smoothieware, or TinyG..
Mar 24 15:33:57 rpi2b-cncjs systemd[1]: /etc/systemd/system/cncjs.service:28: Standard output type syslog is obsolete, automatically updating to journal. Please update your unit file, and consider removing the setting altogether.
Mar 24 15:33:57 rpi2b-cncjs systemd[1]: /etc/systemd/system/cncjs.service:29: Standard output type syslog is obsolete, automatically updating to journal. Please update your unit file, and consider removing the setting altogether.
  - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
  [i]  CNCjs is now started and can be accessed at: ( http://localhost:80 ) | ( http://192.168.1.46:80 )

  Finished
  [i]  Script Finished, its recommended that you reboot your Raspberry Pi.
   └── To reboot run the command: sudo reboot
```

cheton, 2023-03-25:
A network error occurred while attempting to fetch the `react` package. Have you tried retrying to see if it was successful?

```
3115 verbose type invalid-json
3116 verbose stack FetchError: invalid json response body at https://registry.npmjs.org/react reason: Unexpected end of JSON input
3116 verbose stack     at /usr/share/nodejs/npm/node_modules/minipass-fetch/lib/body.js:77:31
3116 verbose stack     at async Promise.all (index 95)
3116 verbose stack     at async Arborist.[buildDepStep] (/usr/share/nodejs/@npmcli/arborist/lib/arborist/build-ideal-tree.js:854:5)
3116 verbose stack     at async Arborist.buildIdealTree (/usr/share/nodejs/@npmcli/arborist/lib/arborist/build-ideal-tree.js:209:7)
3116 verbose stack     at async Arborist.reify (/usr/share/nodejs/@npmcli/arborist/lib/arborist/reify.js:127:5)
3116 verbose stack     at async install (/usr/share/nodejs/npm/lib/install.js:38:3)
3117 verbose cwd /home/pi
3118 verbose Linux 6.1.19-v7+
3119 verbose argv "/usr/bin/node" "/usr/bin/npm" "install" "-g" "cncjs@1.10.1" "--unsafe-perm"
3120 verbose node v12.22.12
3121 verbose npm  v7.5.2
3122 error code FETCH_ERROR
3123 error errno FETCH_ERROR
3124 error invalid json response body at https://registry.npmjs.org/react reason: Unexpected end of JSON input
```

Cjkeenan, 2023-03-25:
Yeah I tried it twice (one after a system reboot). I can try again tomorrow though.

Cjkeenan, 2023-03-26:
I just tried it again, and the same issue occurred. Not a big deal, but still a bit weird. I would upgrade my 2B if there was any stock of RPi's these days.

AustinSaintAubin, 2024-07-11:
Updated the [cncjs-pi-raspbian](https://github.com/cncjs/cncjs-pi-raspbian) install script.
The [Lite x64](https://downloads.raspberrypi.org/raspios_lite_arm64_latest) edition is now recommended.
Recommended RP4 or greater hardware. 

Works with:
- NodeJS: v18.19.0
- NPM: v9.2.0

closed 2024-07-11 by AustinSaintAubin