cncjs/cncjs-pi-raspbian #33: OS is not compitable

kozo2852, 2023-08-24:
I have been having issues trying to install CNCjs on a new Pi 4B. My intentions is to use a 10.1" touchscreen mounted in the panel door as a stand alone controller for my CNC router.

I was getting errors with the traditional install from the CNCjs instructions. Searching the issues I found your installer.

I have tried a few different PI OS versions, and went to the "legacy" version this last time. When I call the installer to run it starts but gives "Detected OS is NOT Compatable with this install script!"

Detected HW: Raspberry Pi 4 Model B Re 1.5

last installed OS: Raspberry Pi OS (Legacy) with desktop
Release date: May 3rd 2023
System: 32-bit
Kernel version: 5.10
Debian version: 10 (buster)
Size: 788MB

Is there a known working PI OS version that works with Pi 4 B?

AustinSaintAubin, 2023-08-24:
I would install the latest main stream x64 Pi OS.

You can alternatively opt to ignore the warnings by unchecking the first option in the install script (ignore compatibility checking). Script might still work.

I will test compatibility later this week, though most recent test a few months ago passed.

kozo2852, 2023-08-24:
Thanks for the follow up! I initially tried the most recent x64 OS but that was before the script. I will try again tonight.