cncjs/cncjs-pi-raspbian #34: command sudo shutdown -h now not working

Ikswoknib, 2024-02-02:
Hi,

I installed cncjs with this script on an Raspberry 3, this worked very well.
Now i would like to add a command to shutdown the RPi. I added a command named Shutdown and used "sudo shutdown -h now" as command.
But if i execute the command from the web frontend nothing happens. In the frontend i can see that it is an errno1 but thats it.
In Syslog I can see this:
`Feb  2 22:20:10 raspberrypi cncjs[468]: 2024-02-02T21:20:10.346Z - #033[32minfo#033[39m api:commands run: title="Shutdown", commands="sudo shutdown -h now"
Feb  2 22:20:10 raspberrypi cncjs[468]: PID:1179> sudo: unable to change to root gid: Operation not permitted
Feb  2 22:20:10 raspberrypi cncjs[468]: PID:1179> sudo: error initializing audit plugin sudoers_audit
`
So it looks like it is an permission problem. But if i log in via SSH and use the same command as user pi it works, so user pi has the permission....

Any Ideas?

perivar, 2024-03-10:
Support Sudo Access
sudo nano /etc/systemd/system/cncjs.service

# Capabilities & Security Settings
#CapabilityBoundingSet=CAP_SYS_ADMIN CAP_NET_BIND_SERVICE  # Commment this out if you want CNCjs to be able to run "sudo" commands.

Restart
systemctl daemon-reload
sudo systemctl stop cncjs.service
sudo systemctl start cncjs.service

Test by running sudo reboot from cncjs gui
See System Log file (syslog)
journalctl -f

Ikswoknib, 2024-03-10:
Thank you, now it works.

Ikswoknib, 2024-03-11:
Thank you, now it workThank you, now it works.
s.

closed 2024-03-11 by Ikswoknib