cncjs/cncjs-pi-raspbian #37: Headless setup screen scale

Elespecia, 2024-07-27:
While running the setup script through SSH (Putty), checkboxes is out of screen to the left.
Also, the latter part of the option description is out of screen to the right.

This is running everything default in Putty as well as doing a Lite installation of Raspberry OS x64.

Normal command line and raspi-config is fine.

AustinSaintAubin, 2024-07-27:
Hi @Elespecia, can you provide a screenshot. 

I have not tested the whiptail UI in Putty. Will try to test this to see if I can reproduce.

I could not find anything immediately related to Putty and whiptail with a Google search.

Elespecia, 2024-07-28:
Sure, here it is.
Usally I run a Docker VM on Proxmox, but Promox have some issued with xchi driver that hasn't been solved so I figured utilizing a **RPI3B** that I had laying around.
[image]

Elespecia, 2024-07-28:
For some reason, when adjusting the window it solved the issue.
I tried that earlier but not wide enough.
So, solution is to make the windows wide enough to fall into the next intervall where the aspect raito matches the information presented in the script.

Might be able to adjust in the script but solvable by Putty.

Problem solved, case closed. Thanks 👍🏻

closed 2024-07-28 by Elespecia