cncjs/cncjs-pi-raspbian #40: Big problems with setup script (it fails with error code 100) + Default port is 80 (and cannot change it)

spanzetta, 2025-04-05:
3 main issues..

1) many errors (code 100, 1 and 2) during setup

2) the default port is 80.. I change to 8000 (see screenshot below) but the script apparently ignore that settings and it setup at port 80 (see logs output).. result is that it doesn't work

3) setup of MJPEG streamer (for webcam, which is connected) fails with error code 1 and 2.. 

See all console output attached

[image]

[Output console.txt](https://github.com/user-attachments/files/19617687/Output.console.txt)

[Output journal.txt](https://github.com/user-attachments/files/19617686/Output.journal.txt)

spanzetta, 2025-04-30:
anyone here?

AustinSaintAubin, 2026-04-25:
Testing install script and updating this weekend. Working to resolve issues like this.